SolderedElectronics/Inkplate-6-hardware #3: Sister project? The Open Book

JustGitting, 2020-02-13:
Hi,

It's exciting to see the Free-Software / Open Hardware movement gain momentum with the explicit aim of protecting our privacy. 

I just found [The Open Book](https://github.com/joeycastillo/The-Open-Book)" project that is currently developing a open-source make it yourself e-reader. The specifications are lower, but the goals are just as high.

1) Would you be able to make a comparison between the Inkplate and The Open Book?

2) Would you consider helping each other as you bother are working on essentially the same project? (I'm not affiliated, I just want both of your projects to succeed!)

Best wishes.

davidzovko, 2020-04-06:
Hey there, sorry for late reply! 

The Open Book project is a great one. As my knowledge goes, it's made as open-source e-book reader. It has all you need to get started with e-book quickly, which Inkplate does not have. On the other hand, Inkplate has a simplicity to get you started with any kind of project which includes e-paper display, comparing to Open Book which is specialised. 
So, it depends on the use case, but I see both product as a great contribution to open source community! 

Of course, I have just sent a suggestion to the Open Book project owner. Let's see what he thinks :) 

Thank you for the suggestion!
David

closed 2020-04-06 by davidzovko

JustGitting, 2020-04-06:
No worries, I understand you and e-radioicacom have been busy!