SolderedElectronics/Inkplate-6-hardware #5: Please make actual source files available

sebastius, 2020-05-03:
Hi, this is not open-source hardware. I love the concept behind Inkplate and i'd love to contribute in the design. Sadly you've not published the actual source files, so i cannot contribute to your work. Please publish your sources, and also publish your iteration commits so we can see the design evolve and help out!

sebastius, 2020-05-09:
Oh another bit: you're not even including the component layout. So When you make these PCBs yourself, you have to reverse engineer the BOM+Schematic to figure out which bit goes where. 

This is not open source hardware. Far from it even.

bbernhard, 2020-12-03:
I would also be interested in the eagle(?) files, in order to do some small modifications. Are there plans to opensource those files too? @davidzovko

dk5ra, 2020-12-05:
also interested! @davidzovko would be lovely

c5e3, 2020-12-05:
where is the pcb-layout source file of your cad software?

ziguana, 2021-05-29:
any updates on this?

SamantazFox, 2021-06-16:
Bump, I'd like to see the sources too!

Please provide Eagle/KiCad schematics and PCB design files, and BOM as CSV, not proprietary XLSX!

davidzovko, 2021-11-09:
Sorry not to publish the original source files. The files which we have are unavailable to be published publicly because of the software error which we have in Altium. We must redraw it in KiCad and afterwards, we will share it.

tiri, 2022-01-21:
I'd also like to see the source files publicly. Will they be released in the near future?

davidzovko, 2022-01-21:
Hi, yes, we are doing a redraw in KiCad and will publish them as soon as they are ready. We estimate about 2 months to get this done.

tiri, 2022-01-21:
Sounds awesome! Are you also going to publish also the 10" Inkplate source files?

davidzovko, 2022-01-22:
We are going to redo all board, incl. 10, in KiCAD and Soldered branding and release them here on our GitHub.

schroedi83, 2022-07-17:
Any update on this?

tommy-hartmann, 2023-01-16:
I bought the Inkplate6 just recently. I also printed the Case and and had to realize that it does not fit.
It would be nice to at least have a note in the Readme that there is a new version pf the board.

I am also interested which GPIO pin is used for the wakup button. Could we at least have a screenshot of the PCB Design?
Thanks!

c5e3, 2023-01-18:
i have the parts since more than 2 years, i think i'll just throw them away

ifadiga, 2023-01-22:
Hello,

Sorry for the problem with the wrong files for 3D model and schematics. 

As David said before we working on new branding and doing all our boards in KiCad. We will be finished some of them for publishing on GitHub.

Next week we will have files for Inkplate 10. For the other Inkplate boards, I will put schematics and 3D models here in the comment until we publish all files.

ifadiga, 2023-01-25:
Hello
Here are the schematics for the four Inkplate that we finished. That are schematics for the new version of Inkplate. We fixed some things on it, but it is similar to the old one. If you need old schematics for some reason, tell me, and I can send you a PDF file. When asked for a schematic, write which GPIO expander you have on board and which branding (e-radionica or Soldered).

[Soldered Inkplate 10 Schematics.pdf](https://github.com/SolderedElectronics/Inkplate-6-hardware/files/10498529/Soldered.Inkplate.10.Schematics.pdf)
[Soldered Inkplate6 Schematics.pdf](https://github.com/SolderedElectronics/Inkplate-6-hardware/files/10498530/Soldered.Inkplate6.Schematics.pdf)
[Soldered Inkplate 6COLOR Schematics.pdf](https://github.com/SolderedElectronics/Inkplate-6-hardware/files/10498531/Soldered.Inkplate.6COLOR.Schematics.pdf)
[Soldered Inkplate 6PLUS Schematics.pdf](https://github.com/SolderedElectronics/Inkplate-6-hardware/files/10498532/Soldered.Inkplate.6PLUS.Schematics.pdf)

We are working on some details for the 3D case, and when finished, I will send it to you.

ifadiga, 2023-02-07:
We published files for our new Inkplate 10 board. You can see them on the [link ](url=https://github.com/SolderedElectronics/Soldered-Inkplate-10-hardware-design)

olalid, 2023-06-02:
I have just printed 3D case from files in the repository only to realise that it is not at all compatible with the board I just bought. The mechanical drawings are also not for the board I bought.
When can we expect correct drawings and/or 3D model to be available?

ifadiga, 2023-06-06:
Hello, 
Sorry for the wrong file for Inkplate. 

We working on the process for updating hardware files on GitHub and I hope that it will be done this week then all files will be available on our GitHub.

rosspointon, 2023-12-20:
@ifadiga - is there any update on the availability of the updated Inkplate 6 case 3d printable design? I just bought a replacement board without an enclosure thinking I could re-use my existing case but the hardware size/layout has changed.

rosspointon, 2023-12-20:
actually this library looks like it has the updated versions - its just quite hidden - https://github.com/SolderedElectronics/Soldered-Inkplate-6-hardware-design/

ifadiga, 2024-03-12:
Hello,

We've recently changed our storage system, and as a result, we don't yet have a repository for Inkplate 5. Due to these changes, I can't determine when the automatic repository setup will function as it did before.

For now, I'll send all the files in a zip file.
[Soldered Inkplate 5.zip](https://github.com/SolderedElectronics/Inkplate-6-hardware/files/14569012/Soldered.Inkplate.5.zip)