SolderedElectronics/Inkplate-6-hardware #7: Wifi on battery power not working

tobias--, 2020-09-09:
I am experiencing issues with my inkplate whenever I try to connect to a wifi and power the inkplate using a LiPo battery (3.7V, 1000mAh). It seems that starting up wifi momentarily draws too much current which triggers the battery protection circuit. When I try to measure the current it seems to peak at 1A, the capacity of the battery. This leads me to believe that powering Wifi might be drawing more the 1A. Is this to be expected by the ESP32?

When I connect power the inkplate using USB on startup and then disconnect USB after a wifi connection has been established, the battery is able to power the board and wifi transmissions just fine. 

To rule out a faulty protection circuit/battery a different battery with larger capacity is already in the mail. However, I am wondering whether this is a known issue, a problem with my inkplate or the way I try to connect to a wifi.

BornaBiro, 2020-09-09:
Yes, ESP32 can draw very high current spikes, while connecting to a WiFi network, especially if you have bad WiFi reception. In fact, in ESP32 datasheet minimal power supply current requirement is 500 mA, which means that ESP32 can draw even more. Ffor some weird reason, WiFi radio while powering up can draw few amps of current (2A peaks in duration of 100uS). Also, there is epaper that itself does not consume any energy while keeping content on the screen, but when it refreshes, it can draw very high current.

In our testing, @3.6V Inkplate was consuming about 0.8A to 1.5A peak while refreshing the screen and connecting to WiFi network, which is more than 1A that you measured, so this is perfectly fine.

We highly advise you to use a good 18650 Lithium battery for Inkplate, or if you really wanna use these small LiPo batteries, you can try to use it with supercap with low ESR in parallel with the battery.

davidzovko, 2020-09-15:
If there are no more comments, we will close this issue.

closed 2020-09-15 by davidzovko

tobias--, 2020-09-16:
Thank you for looking into this!

I neither had luck with a new 2000mAh LiPo battery. It seems to connect briefly to a WiFi but immediately restarts. And I did not even draw anything on the display. Is there any way to completely turn off the display (or other components on the board) in order to separate WiFi and display action?

1.5A peak seems high for a device that is based on ESP32. Do you know what causes this and are there any plans on optimizing Inkplate to be run on battery power?

You recommend a specific LiPo battery for inkplate with 1200mAh. But at the moment I doubt that it will work in combination with WiFi. Maybe you could mention this in the documentation somewhere.

davidzovko, 2020-09-26:
All our testing was done using the [mentioned](https://e-radionica.com/en/li-ion-battery-1200mah.html) 1200mAh battery. It is not everything in the capacity, some batteries are able to source larger currents comparing to others. Thus, 1200mAh might work while 2000mAh might now.

But we would definitely would like to look more deeply into this issue. Can you give us a link to where you have bought your 2000mAh battery? 
Could you share the code you are using? 

We are definitely looking to reduce "power-hungry" peaks that Inkplate has and providing that info will be helpful in reaching that.