SolderedElectronics/Inkplate-6-hardware #10: Powers up fine from USB, but not from battery

jf1452, 2020-11-02:
I'm working on an OpenWeather display based on the code you provided as an example.

I have tried a smaller battery (the same part number that you supply), and the much larger battery in the photo which was salvaged from a tablet with a broken screen. But the unit will not start on battery alone; it runs fine from USB.

[image]

The trace above is at the connector on the Inkplate, you can see the nominal 4.2V rail which has disturbances to 4.04V (probably when starting the WiFi; nothing jumps out as a major concern here. I have also probed the 3V3 rail on the Inkplate and that also stays steady.

[image]

The image is of the battery; although used, it still has good charge capacity and able to drive a 1A load.

For interest purpose, here's a photo of my OpenWeather display; still more work to be done before I release the code, but I have uploaded what I've done so far to git.

[image]

Any idea on what the issue is and why I cannot power InkPlate from a battery?

BornaBiro, 2020-11-04:
Hello jf1452!

I'm sorry to hear that you have issues with your Inkplate.

We investigated this issue and it turns out that it is not a power issue, rather it's an issue related to reset IC. Reset IC is responsible for proper power up and power down of Inkplate. This IC has a manual reset input, which is connected to the reset pushbutton. For some reason, when Inkplate is powered from the battery there is enough noise on trace that connects reset pushbutton to the reset IC MR pin causing reset IC to reset whole board.
There are two solutions to this problem:
 - Connect 10k - 100k pull up resistor on reset IC MR pin (reset IC is small 5 pin IC near epaper connector and MR (manual reset) is 4th pin on this IC. Pull up should be connected to 3.3V rail.
 - There is a small SMD 100nF capacitor near reset pushbutton. Add 2.2 - 4.7uF capacitor parallel to it.

davidzovko, 2020-11-05:
Correct, as Borna suggested, with photo: http://forum.e-radionica.com/en/viewtopic.php?f=22&t=288#p274

We will fix this in the next HW revision.

jf1452, 2020-11-05:
Yes, the extra capacitor and / or the resistor fixes the issue.

As an extra note, when powered from USB, I also experienced frequent resets, originally I thought I had a software issue but it has also now been resolved so I suspect MR was being applied from any power source.

closed 2020-11-05 by jf1452

Zylanx, 2023-11-05:
> Correct, as Borna suggested, with photo: http://forum.e-radionica.com/en/viewtopic.php?f=22&t=288#p274
> 
> We will fix this in the next HW revision.

The forum link is dead, is there any information in the fix?