SolderedElectronics/Inkplate-6-hardware #13: Having trouble getting battery to charge

vcazan, 2021-11-18:
Hello, I have 3 inkplate devices here with standard off the shelf 3.7v li-ion batteries plugged in and I cannot get any of them to charge. Is there any special software definitions that must be set for the charger to be enabled? The LED charging light does not turn on when the micro usb cable is plugged in to the board. Thanks in advance for the help.

davidzovko, 2021-11-18:
Hey, which battery do you use? 

Did you confirm the polarity of the battery matches the polarity on the board connector?

vcazan, 2021-11-18:
Using this battery currently and I can confirm the polarity of the connector is correct.
https://www.amazon.ca/gp/product/B091YMWC72/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

BornaBiro, 2021-11-22:
Can you check if there is any voltage on Inkplate JST connector when there is no battery connected and Inkplate board is connected to the USB power supply?

Can you also check if there is any power on Inkplate board?

ADeadPixel, 2022-07-12:
I am having the same issue with my Inkplate 6PLUS. I do measure voltage across the JST connector (~4.82v usb only & ~3.81v batt plugged in) when it is plugged in via usb. I have verified that the polarity is correct (I needed to flip my battery wires). 

What is odd is when I test the charge LED for voltage (~.3v usb only & ~.4v batt plugged in) it very briefly and very dimly lights up. Now when I test for continuity and touch both sides of the charge LED it lights up very brightly (~1.78v usb only & ~1.78v batt plugged in). This leads me to think that maybe the trace or connection is faulty or something else is going on.