SolderedElectronics/Inkplate-6-hardware #15: LED Charge not working -> question

maltejahn, 2022-04-12:
Hi,

the LED charge indicator doenst do anything. If i am right, the little green dot of the LED should face "away" from the resistor. But on m bought pcb the dot faces towards the 1k resistor. -> Inkplate 6
Ok, you cant fix this from far away.

But, i also have trouble to read out the "nearly" correct voltage. If I use your example Inkplate_Battery_Voltage_And_Temperature.ino I get for e.g. 4.18V at the connector of the LiPO a shown voltage of 4.61 V. I read about a software bug related to the mosfet. But I think this should already be implemented since this problem was fixed in late 2020?

I now would like to get the raw ad values to correct it myself. But i wasnt able as the needed functions are implemented in the inkplate library. Is it possible to add a "double System::readBattery_raw() function in the system.cpp. I could do it on my own, but with the next update this will be probalby gone.

And one last question. What is the minimum Voltage, measured at the battery connector, the Display will work. The ESP should work at a minimum voltage of 3.0 V. The Texas LDO has a drop of 280 mV @ 250 mA. 
So, the Display should work fine till the battery goes down to about 3.3 V? I am looking where i should set my "go charge me" Information

BornaBiro, 2022-04-13:
Hello maltejahn!

LED is properly rotated on all our Inkplate boards as far as we know. What can happen is that LED in the reel can be rotated the opposite way and really we can't do much about it (because the pick and place machine doesn't look for LED orientation on the camera). We are really sorry about that.

About the issue with reading battery voltage; it's not a problem with the MOSFET switch, it's the inaccuracy of ESP32 ADC. There are many topics about that issue on esp32 forum and their github page like [this](https://esp32.com/viewtopic.php?t=13089) and [this](https://github.com/espressif/arduino-esp32/issues/1804). Long story short, every Inkplate board needs to be [calibrated](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/examples/Inkplate6/Advanced_Inkplate_Features/Inkplate_Battery_Voltage_And_Temperature/Inkplate_Battery_Voltage_And_Temperature.ino#L44).

For RAW ADC readings, you can see how [readBattery() functions](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/include/System.cpp#L102) works and you can implement this ADC RAW function yourself outside the Inkplate library so it doesn't break or disappear in the next library update. Just before reading, enable the MOSFETs for the voltage divider, measure it and then disable the voltage divider. :)

Inkplate should work down to 3.1V in out testing (with a good battery that can supply at least 1A of current) without the WIFI, and with the WiFi on is 3.3V.

maltejahn, 2022-04-13:
Hi,

and thanks for you reply. 

I  already tried to do my on raw reading, but im stuck with the way to do it:

/include/myADC.h
`#include "Arduino.h"
#include "SPI.h"
#include "time.h"
#include "Esp.h"
#include "Wire.h"

#include "../.pio/libdeps/esp32/InkplateLibrary/src/include/defines.h"
#include "../.pio/libdeps/esp32/InkplateLibrary/src/include/System.h"

#include "../.pio/libdeps/esp32/InkplateLibrary/src/include/Mcp.h"

double myReadBattery();`

/src/myADC.cpp
`#include "../include/myADC.h"
double myReadBattery()
{
    pinModeInternal(MCP23017_INT_ADDR, mcpRegsInt, 9, INPUT);
    return (double(1));
    
}`

I only added these lines, because i get the error that pinModeInternal wasnt found.

maltejahn, 2022-04-24:
I close this because it's related to the library usage.

I wasn't able to get the myADC function out of system.cpp without errors. So the system.cpp is still modified and will be overwritten when I update all files.

But now I did some measurements (ad raw value - > corresponding voltage) and it fits quite well in the range of interest.

Some suggestions for an hardware revision:
- a jumper and two solder pads to get a ampere meter in line with the battery
- solder pads for the status led to be able to place it anywhere
- or be able to connect the status pin to the mcp to read if it's charging or done

closed 2022-04-24 by maltejahn